MEGA65/mega65-core #876: STQ instruction, mismatched byte order when writing to mapped palette bank registers ($D100-$D3FF)

archiecrux, 2025-03-03:
Platform: MEGA65R3
Core Commit: 3C10488 (release-0.96)
ROM Release: 920395

Executing an STQ instruction on mapped colour palette ($D100-$D3FF) results in a byte order mismatch.
Normally, CPU registers are stored in AXYZ order. In this particular case, register A is omitted and register Z is written twice, resulting in byte order XYZZ.

e.g.

    LDA #$11
    LDX #$22
    LDY #$33
    LDZ #$44
    STQ $D100

which results in the written byte order $22 $33 $44 $44

archiecrux, 2025-03-03:
Small addendum:

- Setting the B register to $D1 makes no difference.
- Addressing VIC registers at $D000 works as expected.

lydon42, 2025-03-08:
Also happens with development.