MEGA65/mega65-core #877: After Etherload, can't change CPU speed with SPEED command

dansanderson, 2025-03-10:
**Test Environment (required)**
You can use MEGAINFO from within FREEZER to obtain this information
(long press RESTORE to enter FREEZER, then press HELP to start MEGAINFO)
 - Platform: MEGA65R6
 - Core Commit: v0.96 (3c10488)
 - ROM Release: 920412

**Describe the bug**
After any use of Etherload to transfer a file to the SD card, to load a ROM, or load a PRG, the BASIC 65 `SPEED` command cannot be used to change the CPU speed.

I would call this a low priority edge case except it's probably worth digging up an explanation, in case there's a larger issue.

**To Reproduce**
Steps to reproduce the behavior:
1. On PC: `mega65_ftp -e -c put "somefile" -c exit` File transfers, system resets.
2. Open Freezer, confirm CPU speed is 40 MHz.
3. On MEGA65: `SPEED 1`
4. Open Freezer.

Expected: CPU speed is now 1 MHz.
Actual: CPU speed is still 40 MHz.

Repeat with step 1 as `etherload -R mega65.rom`. Same result.

Press Reset button to do a full warm boot with the ROM from the SD card. `SPEED 1` now takes effect as expected.

**Additional context**
I noticed this while testing changes to the `SPEED` command, sending test ROMs via etherload. If that's the only affected use case, I just got unlucky. Ideally any use of etherload shouldn't get the state stuck like this.

ki-bo, 2025-03-17:
I believe this is a problem with mega65_ftp and is not a Core problem. I guess this issue should be moved to the tools repository.

dansanderson, 2025-03-17:
Moved to: https://github.com/MEGA65/mega65-tools/issues/217

closed 2025-03-17 by dansanderson