MEGA65/mega65-core #881: Entering the FREEZER menu, slows down ETHERLOAD transfers.

drex69, 2025-03-24:
**Test Environment**

 - Platform : [ MEGA65 R6 ]
 - Core Commit : [ dev core  - 9EE0739 ]
 - ROM Release : [ dev rom - 920413 ]

**Describe the bug**
When entering the FREEZER menu, and then leaving it, ETHERLOAD transfers take 10-15 seconds to execute.

**To reproduce**
Enter the FREEZER menu, then press 'F3' to resume.
Send a .PRG via ETHERLOAD to the MEGA65.
The disk drive then engages, making a sound, and displays the activity light.
The MEGA65 stays on its default ready screen for about 10-15 seconds, before the .PRG is executed.

ki-bo, 2025-03-27:
I suspect what you see is the ROM accessing the real floppy drive and that takes a longer time to detect it can't read any sectors. I will check why the FREEZER is switching to real floppy drive being selected.

drex69, 2025-05-28:
Just an update....

After installing the 0.97 release core, with ROM 920395, everything is working fine.

After installing #154 development core, with ROM 920416, everything is working fine.

I am not sure what fixed it, maybe a newer version of, 'ETHLOAD' \ 'FREEZER', just speculating.