MEGA65/mega65-core #888: matrix_to_ascii.vhdl: Ctrl + `=` is reporting PETSCII $1f

dansanderson, 2025-04-18:
**Test Environment (required)**
You can use MEGAINFO from within FREEZER to obtain this information
(long press RESTORE to enter FREEZER, then press HELP to start MEGAINFO)
 - Platform: MEGA65R6
 - Core Commit: e3095d7
 - ROM Release: 920413

**Describe the bug**
Pressing Ctrl + `=` from the BASIC editor turns the cursor blue, which looks like a serious problem on the blue background. It's actually a very simple fix: matrix_to_ascii.vhdl matrix_petscii_control key code 53 is returning PETSCII code $1f:
```
    53 => x"1f",
```

This should be:
```
    53 => x"ff",
```

such that Ctrl+`=` does not register a PETSCII typing event. This is consistent with Ctrl + the other ASCII substitute keys, e.g. Ctrl + ',' (comma).

**Additional context**
The typing event queue returns PETSCII $3d `=` when the `=` key is unmodified or shifted, and returns PETSCII $5f backarrow when Mega is held to provide access to an ASCII underscore at this code point. Either of these might be alternatives to "no key" $ff, but "no key" would be most consistent with similar keys in this situation. I'm pretty sure $ff is what was intended, and it was mistyped.

CTalkobt, 2025-04-18:
Rather than return $ff as no key suggest to return the normal = value so that apps if they want can discern Ctrl = sequences.

dansanderson, 2025-04-18:
There is no PETSCII interpretation for the typing event Ctrl+`=`, just as there is no PETSCII interpretation for the typing events Ctrl+`:` or Ctrl+`;`. A program that needs to read the non-PETSCII typing evnets needs to use something other than PETSCIIKEY. In this case, ASCIIKEY + MODKEY would be a supported way to interpret Ctrl+'='.

If we want to interpret unrecognized Ctrl chars as unmodified in PETSCIIKEY, we should do it consistently across multiple key codes. This is possible, but we're not doing it currently, so no-key is the smallest consistent change.

Incidentally, I just noticed that Ctrl+`=` *does* emit $1f using the vintage C65 ROM keyboard scanner. I still think no-key is a better choice: it's just as confusing in the C65 ROM as it is currently with the PETSCII table. 😅

closed 2025-07-20 by lydon42