MEGA65/mega65-core #890: More thorough support for SD card standards

dansanderson, 2025-04-27:
**Is your feature request related to a problem? Please describe.**
Our current SD card driver supports a subset of the SD card specification. It is sometimes difficult to find compatible SD cards for the MEGA65, especially for high speed applications such as Mirage's full-motion video demos.

**Describe the solution you'd like**
Either revise our implementation to comply with more of the spec, or replace it with a more mature third-party open source VHDL/Verilog implementation.

**Additional context**
From a conversation on Discord with Mirage, AmokPhase101, lydon, and deft on April 27, 2025 (starts [here](https://discord.com/channels/719326990221574164/1362146277021712434/1366011588762538014), roughly):

* SD cards bundled with MEGA65 are too slow for these demos: SanDisk Ultra SDHC circle-10 16 GB.
* SanDisk Extreme SDHC-I circle-10 U3 16 GB, labeled 90 MB/s, works.
* SanDisk Extreme PRO SDHC-I circle-10 U3 V30 32 GB, labeled 100 MB/s, does not work. (I assume FMV demo glitches after a few seconds, like most too-slow cards, but is not mentioned.)
* SanDisk Ultra SDHC-I circle-10 U1 32 GB, labeled 120 MB/s, works.
* SanDisk Extreme microSDXC UHS-I U3 V30 128 GB is not supported by the MEGA65 SD card utility (fdisk), gets stuck at "Clearing file system data structures." (We do tell people SDXC is not supported.)

[deft](https://discord.com/channels/719326990221574164/1362146277021712434/1366022037969047653):
> Reminder. Missing a proper SD card implementation for several reasons we rely on legacy/ compatibility mode when accessing the card(s). So whatever SanDisk tells about speed on the package, it is no indication on the actual performance in legacy mode. It is totally possible that a modern theoretically super fast card will perform badly in legacy mode. Having a mega65 based test tool would be really helpful.

* Some discussion on non-SPI mode support being potentially useful.

Via [a genspark.ai discussion](https://www.genspark.ai/spark?id=efc83d55-dd22-44c5-9bbe-1c5099b88ce4) (AI generated):
> The SD card interface has two main operating modes: SPI mode (which is simpler but slower) and native SD mode (which includes 1-bit and 4-bit variants, offering higher performance). While SPI mode is freely implementable, the native SD mode has been subject to licensing requirements from the SD Association.

Open source implementations:
* [VDHL-SD-Library by Simon-77](https://github.com/simon-77/VDHL-SD-Library)
* [Wishbone SD Card Controller (OpenCores)](https://opencores.org/projects/sd_card_controller)
* [ZipCPU SDIO Controller](https://github.com/ZipCPU/sdspi); [About ZipCPU](https://zipcpu.com/about/zipcpu.html)

dansanderson, 2025-04-27:
Note that this is unrelated to our FAT32 filesystem implementation, which is also incomplete in notable ways (e.g. lack of support for a file stored in discontiguous blocks aka "fragmentation").

paich64, 2025-04-27:
Another potential option : https://opencores.org/projects/sdcard_mass_storage_controller

paich64, 2025-04-27:
adding the implementation made @MJoergen  https://github.com/MJoergen/SDCard