MEGA65/mega65-core #895: Automatic core select not working with some C64 cartridges

deftmega, 2025-05-15:
**Test Environment**
You can use MEGAINFO from within FREEZER to obtain this information
(long press RESTORE to enter FREEZER, then press HELP to start MEGAINFO)
 - Platform: r6
 - Core Commit: 20250422.09.aaf4542
 - ROM Release: ROM not involved

**Bug**
MSSIAH and MAGIC FORMEL cartridges (C64 carts) do not trigger automatic core selection

**To Reproduce**
Steps to reproduce the behavior:
1. Switch OFF MEGA65
2. Insert cartridge
3. Switch ON MEGA65
4. See _no_ blue LED
5. C64 core was not triggered
6. See error

**Expected behavior**
C64 core should autoboot. MSSIAH runs if C64 core is manually started.

**Additional context**
Why did I exactly borrow the MSSIAH cartridge to you @lydon42 ?! :D

lydon42, 2025-05-16:
Because so I could determine if it works like a cartridge. Answer: it does not.

Please provide extra debug: enter MEGAFLASH, press HELP for MEGAINFO, press F1 for cartridge debug output. Paste here.

Expected output is $d67e 60 (now 60), which means that neither EXROM nor GAME were low. I also don't expect a signature in 8004, as the MSSIAH wants to be tickled the right way.

So: we won't support that, because it makes no sense to make a special exception for one cartridge nobody actually uses.

lydon42, 2025-05-16:
This is a duplicate of #161

closed 2025-05-16 by lydon42